geoffdavis/esphome-mitsubishiheatpump #135: Numbered fan speed modes?

Arn0uDz, 2024-01-19:
I've been using this project for a while now but i'm still bothered by the fan modes in Home Assistant. When I was still using the Melcloud integration I could select the normal fan modes for my unit: 1 , 2 , 3 , 4 and 5

But since using this project i'm forced to use names instead of numbers for the fan modes. The main problem is that my unit has 5 modes and the names for each mode is so confusing. If you only have 3 modes then low/mid/high is logical, but when you have 5 modes it gets confusing.

These are my settings:
fan_mode: [AUTO, DIFFUSE, LOW, MEDIUM, MIDDLE, HIGH]

This gives me the following options in Home Assistant, in a random order:
-Auto
-Diffuse = 1
-Low = 2
-Medium = 3
-Middle = 4
-High = 5

I keep getting confused with the medium and middle setting. I tried this project "https://github.com/gysmo38/mitsubishi2MQTT", that does have a numbered fan modes but it wasn't stable for me. 

Is there a way to rename them, so I have fan modes with numbers again?

davidcd13, 2024-02-16:
Yes I don't understand why it's not a simple numbering... The medium and middle make no sense...

echavet, 2024-03-16:
fan_map in the swiCago lib implementation is:

`static const char* FAN_MAP[6] = { "AUTO", "QUIET", "1", "2", "3", "4" };`

ESPHome Climate component define this :
```c++
enum ClimateFanMode : uint8_t {
  /// The fan mode is set to On
  CLIMATE_FAN_ON = 0,
  /// The fan mode is set to Off
  CLIMATE_FAN_OFF = 1,
  /// The fan mode is set to Auto
  CLIMATE_FAN_AUTO = 2,
  /// The fan mode is set to Low
  CLIMATE_FAN_LOW = 3,
  /// The fan mode is set to Medium
  CLIMATE_FAN_MEDIUM = 4,
  /// The fan mode is set to High
  CLIMATE_FAN_HIGH = 5,
  /// The fan mode is set to Middle
  CLIMATE_FAN_MIDDLE = 6,
  /// The fan mode is set to Focus
  CLIMATE_FAN_FOCUS = 7,
  /// The fan mode is set to Diffuse
  CLIMATE_FAN_DIFFUSE = 8,
  /// The fan mode is set to Quiet
  CLIMATE_FAN_QUIET = 9,
};
```

A component that does not use the UI Climate Component is free to define any mapping for these values, but I'm not sure there is a way to override the native ESPHome climate component labels in this implementation. At least I didn't find a way in my implementation. So I had  to define an arbitrary mapping for these names.
It looks like this:

```c++
void CN105Climate::controlFan() {

    switch (this->fan_mode.value()) {
    case climate::CLIMATE_FAN_OFF:
        this->setPowerSetting("OFF");
        break;
    case climate::CLIMATE_FAN_QUIET:
        this->setFanSpeed("QUIET");
        break;
    case climate::CLIMATE_FAN_DIFFUSE:
        this->setFanSpeed("QUIET");
        break;
    case climate::CLIMATE_FAN_LOW:
        this->setFanSpeed("1");
        break;
    case climate::CLIMATE_FAN_MEDIUM:
        this->setFanSpeed("2");
        break;
    case climate::CLIMATE_FAN_MIDDLE:
        this->setFanSpeed("3");
        break;
    case climate::CLIMATE_FAN_HIGH:
        this->setFanSpeed("4");
        break;
    case climate::CLIMATE_FAN_ON:
    case climate::CLIMATE_FAN_AUTO:
    default:
        this->setFanSpeed("AUTO");
        break;
    }
}

```

rudizl, 2024-03-16:
Where exactly you define those names? 
BTW I just switched to your component and it works perfectly for now. (FH50VE)

echavet, 2024-03-16:
Actually, you can define it making something like that in your yaml; this is from another component version but the support section is the same:

```yaml
climate:
  - platform: cn105  
    name: ${friendly_name}
    id: "esp32_clim"
    compressor_frequency_sensor:
      name: Compressor frequency (clim Sejour)    
    vertical_vane_select:
      name: Orientation de la Vane Verticale
    # horizontal_vane_select:
    #   name: Orientation de la Vane Horizontale
    isee_sensor:
      name: ISEE Sensor
    auto_sub_mode_sensor:
      name: auto-submode sensor
    sub_mode_sensor:
      name: submode sensor
    stage_sensor:
      name: stage sensor      
    remote_temperature_timeout: 15min
    update_interval: 2s         # shouldn't be less than 1 second
    debounce_delay : 100ms       # delay to prevent bouncing 
    supports:
      mode: [COOL, HEAT, FAN_ONLY, DRY]
      fan_mode: [AUTO, QUIET, LOW, MEDIUM, HIGH]
      swing_mode: ["OFF", VERTICAL]
    visual:
      min_temperature: 17
      max_temperature: 23
      temperature_step: 1.0
```

Arn0uDz, 2024-04-11:
> Actually, you can define it making something like that in your yaml; this is from another component version but the support section is the same:
>
> ```yaml
> climate:
>   - platform: cn105
>     name: [...]

How exactly does this change anything? Those are still the same fan modes i'm complaining about.

Arn0uDz, 2024-04-20:
Solved by installing a custom HACS component and making a climate template.

https://github.com/litinoveweedle/hass-template-climate

closed 2024-04-20 by Arn0uDz