geoffdavis/esphome-mitsubishiheatpump #136: swing mode 1, 2, 3, 4, 5

Enjoy-Combi, 2024-01-20:
Hi,

My AC supports position of the vane. Is-it possible modify the code to support them ?

cameroncros, 2024-01-30:
@Enjoy-Combi I am not related to this project, but I was wondering the same thing, so I did some digging. The problem is that esphome does not support the vane positions. It can support Left-Right swinging as well as up and down though, so the code could be modified to support that.

Enjoy-Combi, 2024-01-30:
Long time but coding, but I will give a try during free time and let you know.

echavet, 2024-03-16:
done here: https://github.com/echavet/MitsubishiCN105ESPHome
This repository is largely based on geoffdavis and swicago's work. But with a redesign.

jamescadd, 2024-03-16:
@geoffdavis I would appreciate it if you gave some direction about whether people should continue to use this repo or migrate elsewhere. Yours continues to work for now and I've used it for ages, but this doesn't seem like a durable path forward if you (understandably) don't have time. 

Maybe best to archive and point folks elsewhere? Sorry if a conclusion was reached in another issue, I haven't kept up on all the conversations. And thank you for all your work here - this has been the single most important HA integration for my house for about 3 years now 🙌

echavet, 2024-03-16:
Hey @jamescadd,
I know Geoff doesn't have much time to maintain this rep. And I think we share the same goals. 
Anyway, I've completed a full redesign of this component, obviously based on the work of Geoffrey and SwiCago. We are only 4  contributors but the repository is open to all willing contributors. I encourage you to take a look. The component has been improved in the sense that it now better respects the constraints of the esphome architecture. This repository will also evolve, as the goal is to fully integrate this component into the ESP home project.
We are considering creating a GitHub organization that will allow us to concentrate all of our work in a single project. But in the meantime, I would be delighted if you could test and provide as much feedback as possible on the new version of this component.

https://github.com/echavet/MitsubishiCN105ESPHome

geoffdavis, 2024-03-17:
I made a request for changes on #109 and I'm waiting for that to happen. I didn't merge that request because of that.

In terms of continuing to use the repository @jamescadd, I don't see what good archiving the project will do unless there's a clear transition path. IMO this belongs merged upstream into ESPHOME but that likely won't happen unless we get a chance to merge the work  @echavet in, as the async threading is likely going to be a problem.

Enjoy-Combi, 2024-03-29:
Hi,

Working good regarding the topic of the post with this version https://github.com/echavet/MitsubishiCN105ESPHome.

closed 2024-03-29 by Enjoy-Combi