geoffdavis/esphome-mitsubishiheatpump #138: how to set control over fan speed from web server?

Grappa2020, 2024-02-08:


BenJamesAndo, 2024-02-18:
I got it working by using code from the [Midea ESPHome YAML](https://community.home-assistant.io/t/midea-branded-ac-s-with-esphome-no-cloud/265236).

```
sensor:
  - platform: template
    name: ${friendly_node_name} fan mode
    id: ${node_id}_fan_mode
    internal: true
    update_interval: 10s
    lambda: !lambda |-
      if (id(${node_id}_my_climate).fan_mode == 2)
      {
        if (id(${node_id}_select_fan_mode).state != "Auto")
        {
          auto call = id(${node_id}_select_fan_mode).make_call();
          call.set_option("Auto");
          call.perform();
        }
      }
      else if (id(${node_id}_my_climate).fan_mode == 3)
      {
        if (id(${node_id}_select_fan_mode).state != "Low")
        {
          auto call = id(${node_id}_select_fan_mode).make_call();
          call.set_option("Low");
          call.perform();
        }
      }
      else if (id(${node_id}_my_climate).fan_mode == 4)
      {
        if (id(${node_id}_select_fan_mode).state != "Medium")
        {
          auto call = id(${node_id}_select_fan_mode).make_call();
          call.set_option("Medium");
          call.perform();
        }
      }
      else if (id(${node_id}_my_climate).fan_mode == 5)
      {
        if (id(${node_id}_select_fan_mode).state != "High")
        {
          auto call = id(${node_id}_select_fan_mode).make_call();
          call.set_option("High");
          call.perform();
        }
      }
      return id(${node_id}_my_climate).fan_mode;

# Select
select:
  - platform: template
    name: "Fan mode"
    id: ${node_id}_select_fan_mode
    icon: mdi:fan
    optimistic: true
    options:
      - Auto
      - Low
      - Medium
      - High
    on_value:
      then:
        - lambda: |-
            auto call = id(${node_id}_my_climate).make_call();
            call.set_fan_mode(x.c_str());
            call.perform();
```

I need to tidy it up first as I had to also add/change the following to get it to work.

```
substitutions:
  node_name: mideahvac # Use a unique name.
  node_id: midea_ac    # Use a unique id.
  friendly_node_name: "Midea AC"
```

```
climate:
    id: ${node_id}_my_climate   # Use a unique id
    name: ${node_name}  
```

RikBast, 2024-02-25:
I tried this, but no luck. A whole lot of messages when flashing to the board. There is no error in the config.
Any specifics I should look for? this is my yaml file:

substitutions:
  name: mitsubishi-slaapkamer
  friendly_name: Split Mitsubishi
  node_name: mideahvac # Use a unique name.
  node_id: midea_ac    # Use a unique id.
  friendly_node_name: "Midea AC"

esphome:
  name: ${name}
  friendly_name: ${friendly_name}
  # Set platform options to use c++ 17 instead of 11
  platformio_options:
    build_unflags:
      - "-std=gnu++11"
    build_flags:
      - "-std=gnu++17"

esp32:
  board: wemos_d1_mini32
  framework:
    type: arduino

# Enable logging
logger:
    # Component esphome.coroutine took a long time for an operation
  logs:
    component: ERROR

# Native API and remote logging
api:
  services:
    - service: set_remote_temperature
      variables:
        temperature: float
      then:
        - lambda: 'id(${node_id}_my_climate).set_remote_temperature(temperature);'

    - service: use_internal_temperature
      then:
        - lambda: 'id(${node_id}_my_climate).set_remote_temperature(0);'

ota:
  password: !secret ota_password

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  power_save_mode: none

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "${friendly_name} Hotspot"
    password: !secret wifi_fallback_ap_password

captive_portal:

web_server:
  port: 80

external_components:
  - source: github://nathanjw/esphome-mitsubishiheatpump@unformat-select

climate:
  - platform: mitsubishi_heatpump
    # name: ${friendly_name}
    # id: hp
    id: ${node_id}_my_climate   # Use a unique id
    name: ${node_name}  

    vertical_vane_select:
      name: Vertical Vane    

    # Optional
    hardware_uart: UART2
    baud_rate: 2400
    update_interval: 500ms

    # Supported features 
    supports:
      mode: ["HEAT_COOL", "COOL", "HEAT", "FAN_ONLY", "DRY"]
      fan_mode: ["AUTO", "LOW", "MEDIUM", "HIGH"]
      swing_mode: ["OFF", "VERTICAL"]
    visual:
      min_temperature: 16
      max_temperature: 31
      temperature_step: 0.5

sensor:
  - platform: template
    name: ${friendly_node_name} fan mode
    id: ${node_id}_fan_mode
    internal: true
    update_interval: 10s
    lambda: !lambda |-
      if (id(${node_id}_my_climate).fan_mode == 2)
      {
        if (id(${node_id}_select_fan_mode).state != "Auto")
        {
          auto call = id(${node_id}_select_fan_mode).make_call();
          call.set_option("Auto");
          call.perform();
        }
      }
      else if (id(${node_id}_my_climate).fan_mode == 3)
      {
        if (id(${node_id}_select_fan_mode).state != "Low")
        {
          auto call = id(${node_id}_select_fan_mode).make_call();
          call.set_option("Low");
          call.perform();
        }
      }
      else if (id(${node_id}_my_climate).fan_mode == 4)
      {
        if (id(${node_id}_select_fan_mode).state != "Medium")
        {
          auto call = id(${node_id}_select_fan_mode).make_call();
          call.set_option("Medium");
          call.perform();
        }
      }
      else if (id(${node_id}_my_climate).fan_mode == 5)
      {
        if (id(${node_id}_select_fan_mode).state != "High")
        {
          auto call = id(${node_id}_select_fan_mode).make_call();
          call.set_option("High");
          call.perform();
        }
      }
      return id(${node_id}_my_climate).fan_mode;

# Select
select:
  - platform: template
    name: "Fan mode"
    id: ${node_id}_select_fan_mode
    icon: mdi:fan
    optimistic: true
    options:
      - Auto
      - Low
      - Medium
      - High
    on_value:
      then:
        - lambda: |-
            auto call = id(${node_id}_my_climate).make_call();
            call.set_fan_mode(x.c_str());
            call.perform();

closed 2025-03-31 by geoffdavis