geoffdavis/esphome-mitsubishiheatpump #139: My device works sometimes but, after shows IDLE.

jesus51998, 2024-02-20:
Hi, can you help me.
I made a lot of changes, for my device work well. the device after install work during an time, more late the device is idle, sometimes it get back and work one moment. The end  the device don't work.

esphome:
  name: $devicename
  friendly_name: $upper_devicename
  
esp8266:
  board: esp01_1m

external_components:
  - source: github://geoffdavis/esphome-mitsubishiheatpump

substitutions:
  devicename: p20240210-2
  upper_devicename: Aire salon 2º #EJEMPLO

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: $upper_devicename
    password: !secret fallback_password2

    
captive_portal:

# Enable logging
logger:
  # ESP8266 only - disable serial port logging, as the HeatPump component
  # needs the sole hardware UART on the ESP8266
  #level: VERBOSE
  baud_rate: 0
#  Elimina 2 lineas de error que salen pero no resuelve el problema
  logs:             
    component: ERROR

# Enable Web server.
web_server:
  port: 80

# Enable Home Assistant API
api:
 
ota:

  # Sync time with Home Assistant.
time:
  - platform: homeassistant
    id: homeassistant_time

climate:
  - platform: mitsubishi_heatpump
    name:  $upper_devicename
# Optional
    update_interval: 500ms
#    baud_rate: 2400
    id: hp
    visual:
      min_temperature: 18
      max_temperature: 28
      temperature_step: 0.5

# Sensors with general information.
sensor:

# Or you could use a HomeAssistant sensor
  - platform: homeassistant
    name: "Temperatura Salon"
    entity_id: sensor.temperature_salon
    on_value:
      then:
       - lambda: 'id(hp).set_remote_temperature(x);'

INFO ESPHome 2023.12.9
INFO Reading configuration /config/esphome/p20240210-2.yaml...
INFO Detected timezone 'Europe/Madrid'
INFO Starting log output from 192.168.1.156 using esphome API
INFO Successfully connected to p20240210-2 @ 192.168.1.156 in 0.014s
INFO Successful handshake with p20240210-2 @ 192.168.1.156 in 0.021s
[17:31:34][I][app:102]: ESPHome version 2023.12.9 compiled on Feb 20 2024, 17:12:57
[17:31:34][C][wifi:573]: WiFi:
[17:31:34][C][wifi:405]:   Local MAC: C8:C9:A3:3E:F9:6E
[17:31:34][C][wifi:410]:   SSID: [redacted]
[17:31:34][C][wifi:411]:   IP Address: 192.168.1.156
[17:31:34][C][wifi:412]:   BSSID: [redacted]
[17:31:34][C][wifi:414]:   Hostname: 'p20240210-2'
[17:31:34][C][wifi:416]:   Signal strength: -54 dB ▂▄▆█
[17:31:34][C][wifi:420]:   Channel: 1
[17:31:34][C][wifi:421]:   Subnet: 255.255.255.0
[17:31:34][C][wifi:422]:   Gateway: 192.168.1.1
[17:31:34][C][wifi:423]:   DNS1: 80.58.61.254
[17:31:34][C][wifi:424]:   DNS2: 80.58.61.250
[17:31:34][C][logger:439]: Logger:
[17:31:34][C][logger:440]:   Level: DEBUG
[17:31:34][C][logger:441]:   Log Baud Rate: 0
[17:31:34][C][logger:443]:   Hardware UART: UART0
[17:31:34][C][logger:447]:   Level for 'component': ERROR
[17:31:34][C][homeassistant.time:010]: Home Assistant Time:
[17:31:34][C][homeassistant.time:011]:   Timezone: 'CET-1CEST,M3.5.0,M10.5.0/3'
[17:31:34][I][MitsubishiHeatPump:062]: ESPHome MitsubishiHeatPump version 2.4.1
[17:31:34][I][MitsubishiHeatPump:504]:   Supports HEAT: YES
[17:31:34][I][MitsubishiHeatPump:505]:   Supports COOL: YES
[17:31:34][I][MitsubishiHeatPump:506]:   Supports AWAY mode: NO
[17:31:34][I][MitsubishiHeatPump:507]:   Saved heat: 22.5
[17:31:34][I][MitsubishiHeatPump:508]:   Saved cool: -1.0
[17:31:34][I][MitsubishiHeatPump:509]:   Saved auto: -1.0
[17:31:34][C][captive_portal:088]: Captive Portal:
[17:31:34][C][web_server:168]: Web Server:
[17:31:34][C][web_server:169]:   Address: p20240210-2.local:80
[17:31:34][C][mdns:115]: mDNS:
[17:31:34][C][mdns:116]:   Hostname: p20240210-2
[17:31:34][C][ota:097]: Over-The-Air Updates:
[17:31:34][C][ota:098]:   Address: p20240210-2.local:8266
[17:31:34][W][ota:106]: Last Boot was an unhandled reset, will proceed to safe mode in 8 restarts
[17:31:36][C][api:139]: API Server:
[17:31:36][C][api:140]:   Address: p20240210-2.local:6053
[17:31:36][C][api:144]:   Using noise encryption: NO
[17:31:36][C][homeassistant.sensor:030]: Homeassistant Sensor 'Temperatura Salon'
[17:31:36][C][homeassistant.sensor:030]:   State Class: ''
[17:31:36][C][homeassistant.sensor:030]:   Unit of Measurement: ''
[17:31:36][C][homeassistant.sensor:030]:   Accuracy Decimals: 1
[17:31:36][C][homeassistant.sensor:031]:   Entity ID: 'sensor.temperature_salon'
[17:31:38][D][api:102]: Accepted 192.168.1.145
[17:31:38][D][api.connection:1121]: Home Assistant 2024.2.1 (192.168.1.145): Connected successfully
[17:31:38][D][time:044]: Synchronized time: 2024-02-20 17:31:38
[17:31:38][D][homeassistant.sensor:024]: 'sensor.temperature_salon': Got state 22.40
[17:31:38][D][sensor:093]: 'Temperatura Salon': Sending state 22.40000  with 1 decimals of accuracy
[17:31:38][D][MitsubishiHeatPump:409]: Setting remote temp: 22.4
[17:31:42][D][climate:396]: 'Aire salon 2º' - Sending state:
[17:31:42][D][climate:399]:   Mode: HEAT
[17:31:42][D][climate:401]:   Action: IDLE
[17:31:42][D][climate:404]:   Fan Mode: AUTO
[17:31:42][D][climate:416]:   Swing Mode: OFF
[17:31:42][D][climate:419]:   Current Temperature: 22.50°C
[17:31:42][D][climate:425]:   Target Temperature: 22.50°C
[17:31:49][D][climate:396]: 'Aire salon 2º' - Sending state:
[17:31:49][D][climate:399]:   Mode: HEAT
[17:31:49][D][climate:401]:   Action: HEATING
[17:31:49][D][climate:404]:   Fan Mode: AUTO
[17:31:49][D][climate:416]:   Swing Mode: OFF
[17:31:49][D][climate:419]:   Current Temperature: 22.50°C
[17:31:49][D][climate:425]:   Target Temperature: 22.50°C

those 3 lines repit sometimes
WARNING p20240210-2 @ 192.168.1.156: Connection error occurred: [Errno 104] Connection reset by peer
INFO Processing unexpected disconnect from ESPHome API for p20240210-2 @ 192.168.1.156
WARNING Disconnected from API

Bye

jesus51998, 2024-02-21:
I have 2 Mitsubishi, the MSZ-AP35VG, works very good with the same program, the MSZ-SF35VE2 doesn't work.

jesus51998, 2024-03-03:
My devices is working, change the software, and push software of "github://echavet/MitsubishiCN105ESPHome" and my MSZ-SF35VE2 work very good.

JaviElio, 2024-06-06:
Same problem here. it works fine for some time, then stops and shows IDLE.
Unplug/plug the external unit returns ESP's to live.
Thanks for the info @iannnnnnnn, I'll try it

closed 2025-03-31 by geoffdavis