geoffdavis/esphome-mitsubishiheatpump #144: Compile error: src/esphome/components/mitsubishi_heatpump/espmhp.cpp:454:46: error: comparison between distinct pointer types 'HardwareSerial*' and 'HWCDC*' lacks a cast

jpgnz, 2024-04-04:
`
Compiling .pioenvs/esphome-web-88c620/src/esphome/components/network/util.cpp.o
In file included from src/esphome/components/api/proto.h:4,
                 from src/esphome/components/api/api_pb2.h:5,
                 from src/esphome/components/api/api_connection.h:4,
                 from src/esphome.h:3,
                 from src/esphome/components/mitsubishi_heatpump/espmhp.h:20,
                 from src/esphome/components/mitsubishi_heatpump/espmhp.cpp:20:
src/esphome/components/mitsubishi_heatpump/espmhp.cpp: In member function 'virtual void MitsubishiHeatPump::setup()':
src/esphome/components/mitsubishi_heatpump/espmhp.cpp:454:46: error: comparison between distinct pointer types 'HardwareSerial*' and 'HWCDC*' lacks a cast
             YESNO(this->get_hw_serial_() == &Serial)
                                              ^~~~~~
src/esphome/core/log.h:97:90: note: in definition of macro 'esph_log_config'
   esp_log_printf_(ESPHOME_LOG_LEVEL_CONFIG, tag, __LINE__, ESPHOME_LOG_FORMAT(format), ##__VA_ARGS__)
                                                                                          ^~~~~~~~~~~
src/esphome/components/mitsubishi_heatpump/espmhp.cpp:449:5: note: in expansion of macro 'ESP_LOGCONFIG'
     ESP_LOGCONFIG(
     ^~~~~~~~~~~~~
src/esphome/components/mitsubishi_heatpump/espmhp.cpp:454:13: note: in expansion of macro 'YESNO'
             YESNO(this->get_hw_serial_() == &Serial)
             ^~~~~
Compiling .pioenvs/esphome-web-88c620/src/esphome/components/ota/ota_backend_arduino_esp32.cpp.o
*** [.pioenvs/esphome-web-88c620/src/esphome/components/mitsubishi_heatpump/espmhp.cpp.o] Error 1
========================= [FAILED] Took 18.77 seconds =========================
`

Fresh HA / esphome 2024.3.1 install.

esphome device config is default with minor adjustments:

`
substitutions:
  name: esphome-web-88c620
  friendly_name: office-heatpump-s-01

esphome:
  name: ${name}
  friendly_name: ${friendly_name}
  name_add_mac_suffix: false
  project:
    name: esphome.web
    version: '1.0'

esp32:
  board: esp32-c3-devkitm-1
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:

# Allow Over-The-Air updates
ota:

# Allow provisioning Wi-Fi via serial
#improv_serial:

wifi:
  # Set up a wifi access point
  ap: {}

# In combination with the `ap` this allows the user
# to provision wifi credentials to the device via WiFi AP.
captive_portal:

dashboard_import:
  package_import_url: github://esphome/firmware/esphome-web/esp32c3.yaml@v2
  import_full_config: true

# Sets up Bluetooth LE (Only on ESP32) to allow the user
# to provision wifi credentials to the device.
esp32_improv:
  authorizer: none

# To have a "next url" for improv serial
web_server:

external_components:
  - source: github://geoffdavis/esphome-mitsubishiheatpump

climate:
  - platform: mitsubishi_heatpump
    name: "${friendly_name}"

    # ESP32 only - change UART0 to UART1 or UART2 and remove the
    # logging:baud_rate above to allow the built-in UART0 to function for
    # logging.
    hardware_uart: UART1
`

jpgnz, 2024-04-04:
I think this is a self fail. For anyone else using the ESP32-C3 (With USB-C), this worked for me:

```
#logger:

climate:
  - platform: mitsubishi_heatpump
    name: "${friendly_name}"
    hardware_uart: UART0
    baud_rate: 2400
```

Disable logging, move to UART0, baud_rate: 2400.

lfdmn, 2024-05-13:
@jpgnz could you explain to me why enabling the logging messes up the COM port type?

I thought the C3 have 2 UARTs to choose from: the CDC one for debugging and another one free to be used

Thanks