geoffdavis/esphome-mitsubishiheatpump #146: ESPHome 2024.4 compilation errors

ztox, 2024-04-17:
I started seeing compilation errors with ESPHome 2024.4. I stripped back the yaml config, and followed the README advice around a clean build.   This appears to possibly be an issue in framework-arduinoespressif8266 but posting here after not finding anything related in a cursory search of other issue reports in https://github.com/esphome/issues/issues. 

### Steps:
Stripped down config yaml to minimum (see attached)

- rm -rf ~/.platformio/
- rm -rf ./.esphome/
- esphome -v compile heatpump-1.yaml

### Configuration files
https://github.com/ztox/homeassistant-config/blob/master/esphome/heatpump-1.yaml
https://github.com/ztox/homeassistant-config/blob/master/esphome/common/packages/heatpump.yaml
https://github.com/ztox/homeassistant-config/blob/master/esphome/common/packages/andover-wifi.yaml

### Excerpt from build output:

`In file included from /Users/tobin/.platformio/packages/framework-arduinoespressif8266/libraries/ESP8266WiFi/src/CertStoreBearSSL.h:26,
                 from /Users/tobin/.platformio/packages/framework-arduinoespressif8266/libraries/ESP8266WiFi/src/WiFiClientSecureBearSSL.h:30,
                 from /Users/tobin/.platformio/packages/framework-arduinoespressif8266/libraries/ESP8266WiFi/src/WiFiClientSecure.h:23,
                 from /Users/tobin/.platformio/packages/framework-arduinoespressif8266/libraries/ESP8266WiFi/src/WiFiServerSecure.h:20,
                 from /Users/tobin/.platformio/packages/framework-arduinoespressif8266/libraries/ESP8266WiFi/src/ESP8266WiFi.h:41,
                 from src/esphome/components/wifi/wifi_component.h:23,
                 from src/esphome.h:58,
                 from src/esphome/components/mitsubishi_heatpump/espmhp.h:20,
                 from src/esphome/components/mitsubishi_heatpump/espmhp.cpp:20:
/Users/tobin/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/FS.h: In static member function 'static time_t fs::FS::_defaultTimeCB()':
/Users/tobin/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/FS.h:260:49: error: reference to 'time' is ambiguous
  260 |     static time_t _defaultTimeCB(void) { return time(NULL); }
      |                                                 ^~~~
In file included from src/esphome/components/homeassistant/time/homeassistant_time.h:4,
                 from src/esphome.h:25,
                 from src/esphome/components/mitsubishi_heatpump/espmhp.h:20,
                 from src/esphome/components/mitsubishi_heatpump/espmhp.cpp:20:
src/esphome/components/time/real_time_clock.h:11:11: note: candidates are: 'namespace esphome::time { }'
   11 | namespace time {
      |           ^~~~
In file included from /Users/tobin/.platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/sys/time.h:414,
                 from /Users/tobin/.platformio/packages/framework-arduinoespressif8266/tools/sdk/lwip2/include/lwipopts.h:3560,
                 from /Users/tobin/.platformio/packages/framework-arduinoespressif8266/tools/sdk/lwip2/include/lwip/opt.h:51,
                 from /Users/tobin/.platformio/packages/framework-arduinoespressif8266/tools/sdk/lwip2/include/lwip/inet.h:43,
                 from src/esphome/components/socket/headers.h:10,
                 from src/esphome/components/socket/socket.h:6,
                 from src/esphome/components/api/api_frame_helper.h:14,
                 from src/esphome/components/api/api_connection.h:3,
                 from src/esphome.h:3,
                 from src/esphome/components/mitsubishi_heatpump/espmhp.h:20,
                 from src/esphome/components/mitsubishi_heatpump/espmhp.cpp:20:
/Users/tobin/.platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/time.h:59:11: note:                 'time_t time(time_t*)'
   59 | time_t    time (time_t *_timer);
      |           ^~~~`

**... build log ...**

`
*** [.pioenvs/heatpump-1/src/esphome/components/mitsubishi_heatpump/espmhp.cpp.o] Error 1
========================= [FAILED] Took 26.98 seconds =========================
`

ztox, 2024-04-18:
Closing this issues, as it is not related to the mitsubishiheatpump package.  Compilation error resolved when I remove the native api time config (andover-wifi.yaml).  

```
time:
  platform: homeassistant
  id: homeassistant_time
```

ztox, 2024-04-18:
Leaving this here for future investigation --
I don't fully understand the dependencies involved between `time`, `climate` and `esphome-mitsubishiheatpump.`  The example below keeps the `time: `configuration, but removes the `climate:` config.  It compiles successfully in 2024.4.0.  

```
# test configuration for isolating ESPHome2024.4 compilation error
# integrated using esp01_1m (8266)

substitutions:
  device_name: heatpump
  friendly_name: "Heatpump 1"

esphome:
  name: ${device_name}
  friendly_name: ${friendly_name}
  area: ${area}

# Generic ESP-01, with 1MB memory (ESP-01S)
esp8266:
  board: esp01_1m
  framework:
    platform_version: espressif8266
    version: recommended

# Disable logging (conflict on 8266 UART)
logger:
  # ESP8266 only - disable serial port logging, as the HeatPump component
  # needs the sole hardware UART on the ESP8266
  baud_rate: 0

## Define common local settings for Andover WIFI and API
ota:
  password: !secret fallback_password

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

api:
# disable encryption for now
#    api_encryption_key: !secret encryption_key

mdns:
  disabled: false

time:
  platform: homeassistant
  id: homeassistant_time

# Mitsubishi Heat Pump Control
external_components:
  - source: github://geoffdavis/esphome-mitsubishiheatpump

# climate:
#   - platform: mitsubishi_heatpump
#     name: "${friendly_name}"

```

markalston, 2024-04-25:
I am having the exact same issue but currently using https://github.com/echavet/MitsubishiCN105ESPHome instead of geoffdavis'
Not sure what the issues would be by removing the time component either.  Not sure if it is needed at all.

To add to the confusion, if you keep the time component and add the captive portal component it all compiles fine again.

Basically it seems that if you remove the captive portal then you also need to remove the time component but this is only true if you have the climate component.  Once you remove the climate component then you can remove the captive portal and/or the time component but with climate you cannot have the time component without the captive portal.

Very weird

markalston, 2024-04-25:
Found another person hitting this issue with emporia vue https://github.com/esphome/issues/issues/5300

> So it seems xtensa time.h is clashing with esphomes time.h or real_time_clock.h. I tried adding #include "time.h" but, I get the same error. I am unsure how to resolve this as I dont use C language.

I agree with their analysis but can't figure out a solution.  Also, I can't figure out why the climate platform triggers this as there are no uses of "time" in the climate platform or geoffdavis' code at all.

markalston, 2024-05-02:
Just adding a note here that running without the time component, as well as without logging, captive portal, web server and ap components has not only not caused any issues for me this past week but seems to have resolved an occasional, random, crash that would knock my esp07's down and stop their communicating with the heat pump.   So, for those in the future hitting this compilation problem or having crashes, try removing all those components and you may find success.

closed 2025-03-31 by geoffdavis