geoffdavis/esphome-mitsubishiheatpump #151: Component esphome.coroutine is marked FAILED

GermanCaruncho, 2024-05-31:
A reference message appears in the log and it does not work. In HA it appears perfectly and there is communication, but the AC split doesn't even flinch. The micro is 8266 and the air is MSZ-DM35VA
Thanks.

syrisara, 2024-06-09:
Same is happening to me with a Wemos D1 v1.
ESPHOME version 2024.5.5
Exact error, [E][component:082]:   Component esphome.coroutine is marked FAILED

bmwilson1978, 2024-06-10:
I also have the same issue, appears to have started this week sometime, but only for one of my units.

[image]

bmwilson1978, 2024-06-10:
> I also have the same issue, appears to have started this week sometime, but only for one of my units. [...]

^^ Just on this, I noticed that one of my HP units was connected to a different SSID to the other which was reporting the issue, changed it to the SSID the other HP unit is using (working fine), and it's not reporting that issue anymore.  This might help @syrisara & @GermanCaruncho ?

syrisara, 2024-06-11:
> ^^ Just on this, I noticed that one of my HP units was connected to a different SSID to the other which was reporting the issue, changed it to the SSID the other HP unit is using (working fine), and [...]

Don't no, switched to

https://github.com/echavet/MitsubishiCN105ESPHome

This is working even better for me.

But thank you for your suggestion!

syrisara, 2024-06-11:
> A reference message appears in the log and it does not work. In HA it appears perfectly and there is communication, but the AC split doesn't even flinch. The micro is 8266 and the air is MSZ-DM35VA Thanks.

You could try
https://github.com/echavet/MitsubishiCN105ESPHome

Working for me.

Brandy-AU, 2024-06-19:
Same issue here, removing the external component referencing this repo stops the error.

What's changed?

wBrhy2, 2024-06-19:
Humm I was about to open a ticket but I'm wondering if this is related. My HVAC has been going into an error mode E5 once or twice per day for a few weeks now. Removing my esp32 from the port physically prevents it from happening. Is this the same?

sanjay900, 2024-06-19:
> Don't no, switched to
>
> https://github.com/echavet/MitsubishiCN105ESPHome
>
> This is working even better for me.
>
> But thank you for your suggestion!

Thanks for pointing this out, it just immediately fixed all my problems!

Marki-Home, 2024-08-13:
[E][component:082]:   Component esphome.coroutine is marked FAILED

Is there a solution?
How?
Thanks

opnordahl, 2024-09-16:
This solved the issue for me: 
 Set "baud_rate: 0" under logger:

# Enable logging
logger:
  logs:
    component: ERROR
  baud_rate: 0

redstorm1, 2024-11-15:
I also have this issue

redstorm1, 2024-11-16:
I've spend a day on this trying different things. 
I have a split system that i put in when i built the house, i ran cat6 to each as it was a new build and i prefer hardwired.
I used WT32-ETH01's and found i didnt need pullups on the data lines.
These 4 units all worked fine up until the other day (running since jan 2022). Noticed one was offline restarted the Unit and all 4 dropped off with the same error 
[image]

The indoor units are:
MSZ-LN25 - Office (currently on a ESP8266)
MSZ-LN3.5 - Guest Bed (recovered after a re install)
MSZ-LN50 - Dining Room (recovered after a re install)
MSZ-LN50 - Living Room (offline with the error)

Here is the weird thing, installing the image over the wire without any changes and the Guest Bed and Dining Room units recovered and booted without the error. the issue remains on the Office and the Living units.

I pulled the office one apart and configured up a nodemcu with pullups on uart0 and the office pump with the ESP8266 booted and connected. The ESP8266 had a 80Mhz frequency was the main difference and WiFi not hardwired.
I Tryed the WT32-ETH01 with pullups and didnt connect. (baud set to 1200, 2400,4800, 9600, 19200, 38400, 57600, 115200 were all tried)
Tried hardware uart2 no connection.
Tried setting the cpu frequency down to 80MHz no connection.
Tried switching to the Mitsubishi CN105 ESPHome component no connection, uart0, uart2, mulitple baud rates tried.
Switch back to the ESP8266 and connected. Changed the component to Mitsubishi CN105 ESPHome and connected, tryed out the energy features, outside temperature etc.

Whats got me scratching my head as 2/4 recovered and have exactly the same config.

[image]
[image]

[image]

A little lost now as to what to try, happy to run some more tests to get these WT32-ETH01s back operating.
One thing i did notice that may be an issue is the 5V from the Unit is measuring 6.7VDC with a multi meter which is quite high, with no level shifters this with pullups is going straight on the TX RX pins of the micro so I wonder if the UART is fried.

hansaplasst, 2025-02-23:
Same issue here. 

Most of the time the issue disappears after a hard- or soft reboot of the ESP device.

This makes me think that there is timing issue regarding a heatpump state which is not yet settled before continuing operation. I'm planning to take a look at the code, but for now I've added a reboot button in the ESPHome config and added this to my dashboard.

Here's how.
Add the following code to the end of the ESPHome yaml file of your heatpump and install this change (OTA).
```
button:
  - platform: restart
    name: "Reboot Device"
    icon: "mdi:restart"

```
To reboot the device goto:
- `Settings - Devices & Services - ESPHome` and  select your heatpump [device](url)
- Under `Configuration` you now see an option to reboot the device

Hopefully this (temporarily) solves your issue. 
If not you should wait for a fix.

closed 2025-03-31 by geoffdavis