geoffdavis/esphome-mitsubishiheatpump #152: esphome 2024.6.0 changed the definition of HARDWARE_UART_TO_SERIAL

sanjay900, 2024-06-19:
Support for more boards than the ESP32/8266 was added, which means we are no longer able to compile

pomah86, 2024-06-20:
I would prefer not to swap, any chance this will be fixed?

sanjay900, 2024-06-20:
Shouldn't be too hard to fix, just need to change that line to the following and import `PLATFORM_ESP8266`
```
serial = HARDWARE_UART_TO_SERIAL[PLATFORM_ESP8266][config[CONF_HARDWARE_UART]]
```

Aresitoo, 2024-06-20:
Got the same problem here.

But dont understand what i need to modify in my code go make it work ( total noob sorry ) 

[image]

ghisch, 2024-06-20:
@sanjay900 made a PR following your recommendation, do not hesitate to review

@Aresitoo can't be fixed from your config file, need to be changed in the code

iddiek, 2024-06-20:
same problem with a esp8266 board
  File "/data/external_components/3b4567cc/components/mitsubishi_heatpump/climate.py", line 66, in to_code
    serial = HARDWARE_UART_TO_SERIAL[config[CONF_HARDWARE_UART]]
             ~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^
KeyError: 'UART0'

Aresitoo, 2024-06-20:
@ghisch So meaning just wait till a fix of the code comes out ? ( sorry I'm zero in code HA and Esphome )

ghisch, 2024-06-20:
@Aresitoo Yes

sanjay900, 2024-06-20:
you could probably just change the external component to point to that PR and see if it works

rkboni, 2024-06-20:
It definitely built, but the first one I updated has not come back online yet (and it's been long enough that I suspect the change either isn't right, or needs more configuration.

Here's my config:

```
substitutions:
  name: "basement-minisplit"
  friendly_name: Basement MiniSplit
  
esphome:
  name: ${name}

esp8266:
  board: d1_mini

# Enable logging
logger:
  # ESP8266 only - disable serial port logging, as the HeatPump component
  # needs the sole hardware UART on the ESP8266
  baud_rate: 0

# Enable Home Assistant API
api:

ota:
  platform: esphome

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  manual_ip:
    static_ip: <static-ip>
    gateway: <gateway-ip>
    subnet: <subnet>
    dns1: <dns-ip>

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "${friendly_name} Fallback"
    password: !secret fallback_password

captive_portal:

# Enable Web server.
web_server:
  port: 80

  # Sync time with Home Assistant.
time:
  - platform: homeassistant
    id: homeassistant_time

# Text sensors with general information.
text_sensor:
  # Expose ESPHome version as sensor.
  - platform: version
    name: ${name} ESPHome Version
  # Expose WiFi information as sensors.
  - platform: wifi_info
    ip_address:
      name: ${name} IP
    ssid:
      name: ${name} SSID
    bssid:
      name: ${name} BSSID

# Sensors with general information.
sensor:
  # Uptime sensor.
  - platform: uptime
    name: ${name} Uptime

  # WiFi Signal sensor.
  - platform: wifi_signal
    name: ${name} WiFi Signal
    update_interval: 60s

switch:
  - platform: restart
    name: "${friendly_name} Restart"

external_components:
  - source: github://ghisch/esphome-mitsubishiheatpump@patch-1

climate:
  - platform: mitsubishi_heatpump
    name: "${friendly_name}"
```

sanjay900, 2024-06-20:
Damn, unfortunately I wouldn't have a clue why that would be

rkboni, 2024-06-20:
> It definitely built, but the first one I updated has not come back online yet, and it's been long enough that I suspect the change either isn't right, or needs more configuration.

Oops, this seems very unrelated... it failed to connect to the Wifi and was sitting in fallback-hotspot mode :/ (which also happened to the prior update to 2024.5.5 a few weeks ago.  It's now connected again and reporting the right data.  Sorry for the unwarranted panic!

sanjay900, 2024-06-20:
Good to hear

ziolelle, 2024-06-21:
> ```yaml
> github://ghisch/esphome-mitsubishiheatpump@patch-1
> ```

This worked perfectly for me, really thank you !!!!

adamderuwe, 2024-06-22:
@pomah86 
> I would prefer not to swap, any chance this will be fixed?

This github repo hasn't had a commit in over 18 months.  I think its probably time to migrate to something that is actively maintained like suggested.  Don't hold your breath that this bug will be fixed in this repo.

geoffdavis, 2024-06-23:
Fixed in #153

closed 2024-06-23 by geoffdavis

pomah86, 2024-06-23:
> Fixed in #153

Updated just now and it all went perfect, thank you :)

lambu76, 2024-06-24:
Hi there,
I executed "Clean Build File" in ESPHome in order to be sure to force the download of last version and now I ahve the following error.

INFO ESPHome 2024.6.1
INFO Reading configuration /config/esphome/wemos-esp32-condizionatore-mansarda.yaml...
WARNING 'wemos_esp32_condiziona_mansarda': Using the '_' (underscore) character in the hostname is discouraged as it can cause problems with some DHCP and local name services. For more information, see https://esphome.io/guides/faq.html#why-shouldn-t-i-use-underscores-in-my-device-name
INFO Detected timezone 'Europe/Rome'
INFO Generating C++ source...
INFO Compiling app...
Processing wemos_esp32_condiziona_mansarda (board: nodemcu-32s; framework: arduino; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
Library Manager: Installing esphome/AsyncTCP-esphome @ 2.1.3
INFO Installing esphome/AsyncTCP-esphome @ 2.1.3
Unpacking  [####################################]  100%
Library Manager: AsyncTCP-esphome@2.1.3 has been installed!
INFO AsyncTCP-esphome@2.1.3 has been installed!
Library Manager: Installing esphome/ESPAsyncWebServer-esphome @ 3.2.2
INFO Installing esphome/ESPAsyncWebServer-esphome @ 3.2.2
Unpacking  [####################################]  100%
Library Manager: ESPAsyncWebServer-esphome@3.2.2 has been installed!
INFO ESPAsyncWebServer-esphome@3.2.2 has been installed!
Library Manager: Resolving dependencies...
INFO Resolving dependencies...
Library Manager: Installing git+https://github.com/SwiCago/HeatPump#5d1e146771d2f458907a855bf9d5d4b9bf5ff033
INFO Installing git+https://github.com/SwiCago/HeatPump#5d1e146771d2f458907a855bf9d5d4b9bf5ff033
git version 2.39.2
Cloning into '/data/cache/platformio/cache/tmp/pkg-installing-h876tpz5'...
HEAD is now at 5d1e146 Merge pull request #209 from dzungpv/master
Library Manager: HeatPump@1.0.0+sha.5d1e146 has been installed!
INFO HeatPump@1.0.0+sha.5d1e146 has been installed!
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Dependency Graph
|-- AsyncTCP-esphome @ 2.1.3
|-- WiFi @ 2.0.0
|-- FS @ 2.0.0
|-- Update @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 3.2.2
|-- DNSServer @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- HeatPump @ 1.0.0+sha.5d1e146
Compiling .pioenvs/wemos_esp32_condiziona_mansarda/src/esphome/components/api/api_connection.cpp.o
Compiling .pioenvs/wemos_esp32_condiziona_mansarda/src/esphome/components/api/api_frame_helper.cpp.o
Compiling .pioenvs/wemos_esp32_condiziona_mansarda/src/esphome/components/api/api_pb2.cpp.o
Compiling .pioenvs/wemos_esp32_condiziona_mansarda/src/esphome/components/api/api_pb2_service.cpp.o
Compiling .pioenvs/wemos_esp32_condiziona_mansarda/src/esphome/components/api/api_server.cpp.o
Compiling .pioenvs/wemos_esp32_condiziona_mansarda/src/esphome/components/api/list_entities.cpp.o
Compiling .pioenvs/wemos_esp32_condiziona_mansarda/src/esphome/components/api/proto.cpp.o
Compiling .pioenvs/wemos_esp32_condiziona_mansarda/src/esphome/components/api/subscribe_state.cpp.o
Compiling .pioenvs/wemos_esp32_condiziona_mansarda/src/esphome/components/api/user_services.cpp.o
Compiling .pioenvs/wemos_esp32_condiziona_mansarda/src/esphome/components/captive_portal/captive_portal.cpp.o
Compiling .pioenvs/wemos_esp32_condiziona_mansarda/src/esphome/components/climate/climate.cpp.o
Compiling .pioenvs/wemos_esp32_condiziona_mansarda/src/esphome/components/climate/climate_mode.cpp.o
Compiling .pioenvs/wemos_esp32_condiziona_mansarda/src/esphome/components/climate/climate_traits.cpp.o
Compiling .pioenvs/wemos_esp32_condiziona_mansarda/src/esphome/components/esp32/core.cpp.o
Compiling .pioenvs/wemos_esp32_condiziona_mansarda/src/esphome/components/esp32/gpio.cpp.o
Compiling .pioenvs/wemos_esp32_condiziona_mansarda/src/esphome/components/esp32/preferences.cpp.o
Compiling .pioenvs/wemos_esp32_condiziona_mansarda/src/esphome/components/esphome/ota/ota_esphome.cpp.o
Compiling .pioenvs/wemos_esp32_condiziona_mansarda/src/esphome/components/homeassistant/time/homeassistant_time.cpp.o
Compiling .pioenvs/wemos_esp32_condiziona_mansarda/src/esphome/components/logger/logger.cpp.o
Compiling .pioenvs/wemos_esp32_condiziona_mansarda/src/esphome/components/logger/logger_esp32.cpp.o
Compiling .pioenvs/wemos_esp32_condiziona_mansarda/src/esphome/components/logger/logger_esp8266.cpp.o
Compiling .pioenvs/wemos_esp32_condiziona_mansarda/src/esphome/components/logger/logger_host.cpp.o
Compiling .pioenvs/wemos_esp32_condiziona_mansarda/src/esphome/components/logger/logger_libretiny.cpp.o
Compiling .pioenvs/wemos_esp32_condiziona_mansarda/src/esphome/components/logger/logger_rp2040.cpp.o
Compiling .pioenvs/wemos_esp32_condiziona_mansarda/src/esphome/components/md5/md5.cpp.o
Compiling .pioenvs/wemos_esp32_condiziona_mansarda/src/esphome/components/mdns/mdns_component.cpp.o
Compiling .pioenvs/wemos_esp32_condiziona_mansarda/src/esphome/components/mdns/mdns_esp32.cpp.o
Compiling .pioenvs/wemos_esp32_condiziona_mansarda/src/esphome/components/mdns/mdns_esp8266.cpp.o
Compiling .pioenvs/wemos_esp32_condiziona_mansarda/src/esphome/components/mdns/mdns_host.cpp.o
[... 99 lines cut ...]
                                             ^~~~~~~~~~~~~~~~~~
src/esphome/components/mitsubishi_heatpump/espmhp.cpp:669:28: error: 'remote_operating_timeout_' was not declared in this scope
         this->operating_ ? remote_operating_timeout_ : remote_idle_timeout_;
                            ^~~~~~~~~~~~~~~~~~~~~~~~~
[... cut ...]
src/esphome/components/mitsubishi_heatpump/espmhp.cpp:671:13: note: suggested alternative: 'enforce_remote_temperature_sensor_timeout'
             last_remote_temperature_sensor_update_.has_value()) {
             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
             enforce_remote_temperature_sensor_timeout
*** [.pioenvs/wemos_esp32_condiziona_mansarda/src/esphome/components/mitsubishi_heatpump/espmhp.cpp.o] Error 1
========================= [FAILED] Took 99.00 seconds =========================

korttoma, 2024-06-24:
> Hi there, I executed "Clean Build File" in ESPHome in order to be sure to force the download of last version and now I ahve the following error.
>
> ## INFO ESPHome 2024.6.1
> INFO Reading configuration [...]

The answer is in your post
 
"https://esphome.io/guides/faq.html#why-shouldn-t-i-use-underscores-in-my-device-name"

lambu76, 2024-06-24:
> The answer is in your post
>
> "https://esphome.io/guides/faq.html#why-shouldn-t-i-use-underscores-in-my-device-name"

It worked fine for years, btw the error doens't seem related to underscore.

Have you seen the error into the compiling log ? Please don't stop to the first rows.

please let me know.

korttoma, 2024-06-24:
Not sure if the errors are related to the hostname or not but I had to change some host names recently to be able to compile and upload so maybe it is wort a try?

btw, what external component are you using? "geoffdavis"? I changed to the "echavet" to work around the "UART" issue.

#external_components:
#  - source: github://geoffdavis/esphome-mitsubishiheatpump
external_components:
  - source: github://echavet/MitsubishiCN105ESPHome

lambu76, 2024-06-24:
> Not sure if the errors are related to the hostname or not but I had to change some host names recently to be able to compile and upload so maybe it is wort a try?
>
> btw, what external component are [...]

I'm using geoffdavis and the error till yesterday was realted to the UART (as in this issue title).

Then today I saw a reply in this thread from geofdavis that the problem was fixed, but now I have that error.

Aresitoo, 2024-06-24:
> Change the external component to this, and you can use the fixed code without waiting for an update :)
> 
> Just grabbed a spare esp32, seems to get past the error just fine
> 
> ```yaml
> external_components:
>   - source: github://ghisch/esphome-mitsubishiheatpump@patch-1
> ```

So I have added this repo, it compiles correctly but now when it wants to push to the board it says 

''ERROR Error auth result: Error: Authentication invalid. Is the password correct?''

if I do not put any line and just

# Enable over-the-air updates.
ota:
  - platform: esphome

I also get error saying it needs a password

my previous config was this

# Enable over-the-air updates.
ota:
  - password: "xxxxxxxxxxxxxxxx" #OPCIONAL
    platform: esphome

tried the new config from esphome website

# Example configuration entry
ota:
  - platform: esphome
    password: !secret ota_password

and put 

## ESPHome
ota_password: xxxxxxxxxxxxxxxx

in the secrets file but is also not working

what am I doing wrong here ?

lambu76, 2024-06-24:
I found the solution in another post in this project.

https://github.com/geoffdavis/esphome-mitsubishiheatpump/issues/117

Basically I added in my esphome below lines.

esphome:
....
  platformio_options:
    build_unflags:
      - "-std=gnu++11"
    build_flags:
      - "-std=gnu++17"  

I kindly ask you to @geoffdavis to advise if this is the behavior expected.

thx

Aresitoo, 2024-06-25:
Ok so in my case I've re-flashed both esp connected to esphome fiscally to the computer, as new projects.

And got everything working and they are updating with the patched external components

Used : 

'''
external_components:
  - source: github://ghisch/esphome-mitsubishiheatpump@patch-1
 '''

Only modification to my old code has been to the OTA part 

'''
ota:
platform: esphome
password: "xxxx"
'''

geoffdavis, 2024-06-28:
> I found the solution in another post in this project.
>
>
>
> https://github.com/geoffdavis/esphome-mitsubishiheatpump/issues/117
>
>
>
> Basically I added in my esphome below lines.
>
>
>
> esphome:
>
> .... [...]

It's not expected behavior but I have been using 8266s to test.

geoffdavis, 2024-06-28:
Note that the echavet repository is a rewrite of this code. I've been working with Eric to try and consolidate repositories, but nothing concrete so far.