geoffdavis/esphome-mitsubishiheatpump #157: Compilation error using 2.5.0 & 2024.6.4

yaro24, 2024-06-27:
With both the current versions of mhp and esphome I can't compile the projects anymore. 

Here's the config:

`
esphome:
  name: klima_wintergarten
  friendly_name: Klima Wintergarten

esp32:
  board: esp32dev
  framework: 
    type: arduino
  
external_components:
  - source: github://geoffdavis/esphome-mitsubishiheatpump
  
# Enable logging
logger:
#  level: DEBUG
#  baud_rate: 0

<api , ota, wifi> 
  
climate:
  - platform: mitsubishi_heatpump
    name: Klimaanlage
    id: hp
    supports:
      mode: [HEAT_COOL, COOL, HEAT, FAN_ONLY]
      #mode: [COOL, HEAT, FAN_ONLY]
      fan_mode: [AUTO, DIFFUSE, LOW, MEDIUM, HIGH]
    hardware_uart: UART2

sensor:
  - platform: wifi_signal
    name: Klima Wintergarten WiFi Signal
    update_interval: 60s
  - platform: homeassistant
    name: "Thermometer"
    entity_id: sensor.berglampe_bmp_temperatur
    internal: true
    filters:
      - throttle: 10s
      - heartbeat: 1min             # Maximum interval between updates.
      - debounce: 1s
    on_value:
      then:
        - lambda: 'id(hp).set_remote_temperature(x);'

`

Here's the log output:

`
INFO ESPHome 2024.6.4
INFO Reading configuration /config/klima-wg.yaml...
WARNING 'klima_wintergarten': Using the '_' (underscore) character in the hostname is discouraged as it can cause problems with some DHCP and local name services. For more information, see https://esphome.io/guides/faq.html#why-shouldn-t-i-use-underscores-in-my-device-name
INFO Generating C++ source...
INFO Compiling app...
Processing klima_wintergarten (board: esp32dev; framework: arduino; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Dependency Graph
|-- AsyncTCP-esphome @ 2.1.3
|-- WiFi @ 2.0.0
|-- FS @ 2.0.0
|-- Update @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 3.2.2
[... 109 lines cut ...]
         this->operating_ ? remote_operating_timeout_ : remote_idle_timeout_;
                            ^~~~~~~~~~~~~~~~~~~~~~~~~
src/esphome/components/mitsubishi_heatpump/espmhp.cpp:669:28: note: suggested alternative: 'set_remote_operating_timeout_minutes'
         this->operating_ ? remote_operating_timeout_ : remote_idle_timeout_;
                            ^~~~~~~~~~~~~~~~~~~~~~~~~
                            set_remote_operating_timeout_minutes
src/esphome/components/mitsubishi_heatpump/espmhp.cpp:669:56: error: 'remote_idle_timeout_' was not declared in this scope
         this->operating_ ? remote_operating_timeout_ : remote_idle_timeout_;
                                                        ^~~~~~~~~~~~~~~~~~~~
src/esphome/components/mitsubishi_heatpump/espmhp.cpp:669:56: note: suggested alternative: 'set_timeout'
         this->operating_ ? remote_operating_timeout_ : remote_idle_timeout_;
                                                        ^~~~~~~~~~~~~~~~~~~~
                                                        set_timeout
src/esphome/components/mitsubishi_heatpump/espmhp.cpp:671:13: error: 'last_remote_temperature_sensor_update_' was not declared in this scope
             last_remote_temperature_sensor_update_.has_value()) {
             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
src/esphome/components/mitsubishi_heatpump/espmhp.cpp:671:13: note: suggested alternative: 'enforce_remote_temperature_sensor_timeout'
             last_remote_temperature_sensor_update_.has_value()) {
             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
             enforce_remote_temperature_sensor_timeout
*** [.pioenvs/klima_wintergarten/src/esphome/components/mitsubishi_heatpump/espmhp.cpp.o] Error 1
========================== [FAILED] Took 4.50 seconds ==========================
`

Please let me know if you need any more information or testing

65698888, 2024-06-29:
Who has the solution to the same problem?

FedericoAntoniazzi, 2024-06-29:
Hello there, 
I pinned the commit before all the merges and confirm it is a regression.

Here for reference:
```yaml
external_components:
  - source: "github://geoffdavis/esphome-mitsubishiheatpump@31ed3068fcf0713873b157e28d0d36248a352d83"
```

TimelessNL, 2024-06-29:
> Here for reference:
> 
> ```yaml
> external_components:
>   - source: "github://geoffdavis/esphome-mitsubishiheatpump@31ed3068fcf0713873b157e28d0d36248a352d83"
> ```

@FedericoAntoniazzi What do you mean exactly?
Using that commit result in another compile error:
```
Compiling .pioenvs/mitsubishi-airco/lib05e/AsyncTCP-esphome/AsyncTCP.cpp.o
src/main.cpp: In function 'void setup()':
src/main.cpp:524:57: error: no matching function for call to 'MitsubishiHeatPump::MitsubishiHeatPump(HWCDC*)'
   mitsubishiheatpump_id = new MitsubishiHeatPump(&Serial);
                                                         ^
In file included from src/esphome.h:48,
                 from src/main.cpp:3:
src/esphome/components/mitsubishi_heatpump/espmhp.h:55:9: note: candidate: 'MitsubishiHeatPump::MitsubishiHeatPump(HardwareSerial*, uint32_t)'
         MitsubishiHeatPump(
         ^~~~~~~~~~~~~~~~~~
src/esphome/components/mitsubishi_heatpump/espmhp.h:55:9: note:   no known conversion for argument 1 from 'HWCDC*' to 'HardwareSerial*'
src/esphome/components/mitsubishi_heatpump/espmhp.h:44:7: note: candidate: 'MitsubishiHeatPump::MitsubishiHeatPump(const MitsubishiHeatPump&)'
 class MitsubishiHeatPump : public PollingComponent, public climate::Climate {
       ^~~~~~~~~~~~~~~~~~
src/esphome/components/mitsubishi_heatpump/espmhp.h:44:7: note:   no known conversion for argument 1 from 'HWCDC*' to 'const MitsubishiHeatPump&'
src/esphome/components/mitsubishi_heatpump/espmhp.h:44:7: note: candidate: 'MitsubishiHeatPump::MitsubishiHeatPump(MitsubishiHeatPump&&)'
src/esphome/components/mitsubishi_heatpump/espmhp.h:44:7: note:   no known conversion for argument 1 from 'HWCDC*' to 'MitsubishiHeatPump&&'
```

jonathandann, 2024-07-11:
```
external_components:
  - source: github://geoffdavis/esphome-mitsubishiheatpump@2.5.0

esphome:
  platformio_options:
    build_flags: 
      - -std=gnu++17
    build_unflags:
      - -std=gnu++11

esp32:
  board: lolin_s2_mini
  variant: esp32s2
  framework:
    type: arduino
```

https://github.com/SwiCago/HeatPump/issues/13#issuecomment-2222657355

SCUZNUTS, 2024-07-14:
> Hello there, I pinned the commit before all the merges and confirm it is a regression.
> 
> Here for reference:
> 
> ```yaml
> external_components:
>   - source: "github://geoffdavis/esphome-mitsubishiheatpump@31ed3068fcf0713873b157e28d0d36248a352d83"
> ```

confirmed this version works for me

adeptcain, 2024-07-20:
> This got it working for me with a Wemos ESP32-S2-mini v1.0.0 https://www.wemos.cc/en/latest/s2/s2_mini.html
> 
> ```
> external_components:
>   - source: github://geoffdavis/esphome-mitsubishiheatpump@2.5.0
> 
> esphome:
>   platformio_options:
>     build_flags: 
>       - -std=gnu++17
>     build_unflags:
>       - -std=gnu++11
> 
> esp32:
>   board: lolin_s2_mini
>   variant: esp32s2
>   framework:
>     type: arduino
> ```
> 
> [SwiCago/HeatPump#13 (comment)](https://github.com/SwiCago/HeatPump/issues/13#issuecomment-2222657355)

Can confirm this worked for me on a cheap clone ESP32. Reverting to a previous commit also worked, but I'd rather be running the latest.

geoffdavis, 2024-07-21:
@adeptcain Thanks for tracking this down. Looks like one of the newer commits starting using C++17 code, and the ESP8266 toolchain is totally fine with it.

@SCUZNUTS and @yaro24 please verify that the "platformio_options" that @adeptcain provided work for your boards, and I'll update the documentation for the ESP32.

SCUZNUTS, 2024-07-22:
Can confirm, doing the above works;

```
external_components:
  - source: github://geoffdavis/esphome-mitsubishiheatpump

esphome:
  platformio_options:
    build_flags: 
      - -std=gnu++17
    build_unflags:
      - -std=gnu++11
```

adeptcain, 2024-07-22:
@geoffdavis thanks, but it was @jonathandann who found the fix by the looks of it, I was just confirming it works for me :)

closed 2025-03-31 by geoffdavis