geoffdavis/esphome-mitsubishiheatpump #165: CN105 problem with network connection

ionutm80, 2024-08-18:
Hi,

I have 3 Mitsubishi units SEZ-M35DA and for each I have already connected a MAC-567IF-E unit on the CN105 port. I have managed to find a CN105 splitter, see picture below so that I can also attach Wemos D1 Mini and control the HVAC locally not through Cloud, however I still need to keep the MAC unit for WAF reasons.
[image]

The problems is that once I attached the Wemos D1 on the same splitter it works and gets connected to both the HVAC and MQTT server but the MAC-567IF-E enters into an error state and losses connection to the internet, I cannot longer control the HVAC through Cloud. 

For somebody who knows the schematics of CN105 port, is it somehow allowing only one connection to the internet to be active?

thecem, 2024-08-18:
Could you pls send hi res pics of the splitter both side of the PCB. If it is possible so good that we could analyse the PCR and parts.

thx!

ionutm80, 2024-08-18:
Is it sufficient?
[image]
[image]

ionutm80, 2024-08-18:
On the flip side are the 2 splitted plastic connection shaped like an U, nothing else.

thecem, 2024-08-18:
The question is what for diodes these are and how the are connected to the cable. For what is this splitter ? To connect two devices to one HP?

ionutm80, 2024-08-18:
I bought it from here:

https://coolautomation-emea.com/product/cn105-splitter-for-me/

thecem, 2024-08-18:
Could you see which diodes these are?

ionutm80, 2024-08-18:
I will ask a professional tomorrow to open the HVAC for me and make better pictures.

thecem, 2024-08-19:
May this could help: since the Aitzone splitter is working either the esp-01  and 5V adapter solution:

https://github.com/echavet/MitsubishiCN105ESPHome/discussions/105#discussioncomment-10375279

ionutm80, 2024-08-21:
Coming to report: last night I have assembled the ESP-01 and the 3.3-5V adapter and connected to the splitter, I have turn-off/on the AC, when back on power the MAC-567IF-E go online and I can cotrol the HVAC through Mitsubishi Cloud while the ESP-01 connects to MQTT but does not connects to the HVAC. I'm already puzzled, the Wemos D1 w/o voltage regulator prevail in fron of the MAC-567IF, meaning it connects to HVAC and put the MAC unit in error state, while the the ESP-01 with voltage regulator is not able to connect and the MAC manages to connect!
The only other possible solution is to try the Wemos D1 with the voltage regulator as proposed here:
https://github.com/unixko/MitsuCon/issues/15#issuecomment-1849293534

thecem, 2024-08-21:
Did you get the ESP-01 runing without the splitter and MAC? 
Try to only connect the ESP-01 (with adapter) to AC.

closed 2025-03-31 by geoffdavis