geoffdavis/esphome-mitsubishiheatpump #167: Compile errors with 2.5 and ESPHome 2024.8.0

redstorm1, 2024-08-25:
[logs_hvac1_run.txt](https://github.com/user-attachments/files/16742359/logs_hvac1_run.txt)

Fails to build with ESPHOme 2024.8.0
see the attached logfile for the build.

monkeypr00f, 2024-08-26:
Same error for me on 2024.8.0, but it failed on 2024.7.3.

`In file included from src/esphome.h:32,
                 from src/esphome/components/mitsubishi_heatpump/espmhp.h:20,
                 from src/esphome/components/mitsubishi_heatpump/espmhp.cpp:21:
src/esphome/components/mitsubishi_heatpump/espmhp.h:182:14: error: 'optional' in namespace 'std' does not name a template type
         std::optional<std::chrono::duration<long long, std::ratio<60>>> remote_operating_timeout_;
              ^~~~~~~~
src/esphome/components/mitsubishi_heatpump/espmhp.h:182:9: note: 'std::optional' is only available from C++17 onwards
         std::optional<std::chrono::duration<long long, std::ratio<60>>> remote_operating_timeout_;
         ^~~
src/esphome/components/mitsubishi_heatpump/espmhp.h:183:14: error: 'optional' in namespace 'std' does not name a template type
         std::optional<std::chrono::duration<long long, std::ratio<60>>> remote_idle_timeout_;
              ^~~~~~~~
src/esphome/components/mitsubishi_heatpump/espmhp.h:183:9: note: 'std::optional' is only available from C++17 onwards
         std::optional<std::chrono::duration<long long, std::ratio<60>>> remote_idle_timeout_;
         ^~~
src/esphome/components/mitsubishi_heatpump/espmhp.h:184:14: error: 'optional' in namespace 'std' does not name a template type
         std::optional<std::chrono::duration<long long, std::ratio<60>>> remote_ping_timeout_;
              ^~~~~~~~
src/esphome/components/mitsubishi_heatpump/espmhp.h:184:9: note: 'std::optional' is only available from C++17 onwards
         std::optional<std::chrono::duration<long long, std::ratio<60>>> remote_ping_timeout_;
         ^~~
src/esphome/components/mitsubishi_heatpump/espmhp.h:185:14: error: 'optional' in namespace 'std' does not name a template type
         std::optional<std::chrono::time_point<std::chrono::steady_clock>> last_remote_temperature_sensor_update_;
              ^~~~~~~~
src/esphome/components/mitsubishi_heatpump/espmhp.h:185:9: note: 'std::optional' is only available from C++17 onwards
         std::optional<std::chrono::time_point<std::chrono::steady_clock>> last_remote_temperature_sensor_update_;
         ^~~
src/esphome/components/mitsubishi_heatpump/espmhp.h:186:14: error: 'optional' in namespace 'std' does not name a template type
         std::optional<std::chrono::time_point<std::chrono::steady_clock>> last_ping_request_;
              ^~~~~~~~
src/esphome/components/mitsubishi_heatpump/espmhp.h:186:9: note: 'std::optional' is only available from C++17 onwards
         std::optional<std::chrono::time_point<std::chrono::steady_clock>> last_ping_request_;
         ^~~
src/esphome/components/mitsubishi_heatpump/espmhp.cpp: In member function 'virtual void MitsubishiHeatPump::control(const esphome::climate::ClimateCall&)':
src/esphome/components/mitsubishi_heatpump/espmhp.cpp:252:9: error: 'last_remote_temperature_sensor_update_' was not declared in this scope
     if (last_remote_temperature_sensor_update_.has_value()) {
         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
src/esphome/components/mitsubishi_heatpump/espmhp.cpp:252:9: note: suggested alternative: 'enforce_remote_temperature_sensor_timeout'
     if (last_remote_temperature_sensor_update_.has_value()) {
         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
         enforce_remote_temperature_sensor_timeout
src/esphome/components/mitsubishi_heatpump/espmhp.cpp: In member function 'void MitsubishiHeatPump::set_remote_temperature(float)':
src/esphome/components/mitsubishi_heatpump/espmhp.cpp:625:9: error: 'last_remote_temperature_sensor_update_' was not declared in this scope
         last_remote_temperature_sensor_update_ =
         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
src/esphome/components/mitsubishi_heatpump/espmhp.cpp:625:9: note: suggested alternative: 'enforce_remote_temperature_sensor_timeout'
         last_remote_temperature_sensor_update_ =
         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
         enforce_remote_temperature_sensor_timeout
src/esphome/components/mitsubishi_heatpump/espmhp.cpp:628:9: error: 'last_remote_temperature_sensor_update_' was not declared in this scope
         last_remote_temperature_sensor_update_.reset();
         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
Compiling .pioenvs/esp32_acmatilde/src/esphome/components/select/select.cpp.o
src/esphome/components/mitsubishi_heatpump/espmhp.cpp:628:9: note: suggested alternative: 'enforce_remote_temperature_sensor_timeout'
         last_remote_temperature_sensor_update_.reset();
         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
         enforce_remote_temperature_sensor_timeout
src/esphome/components/mitsubishi_heatpump/espmhp.cpp: In member function 'void MitsubishiHeatPump::ping()':
src/esphome/components/mitsubishi_heatpump/espmhp.cpp:636:5: error: 'last_ping_request_' was not declared in this scope
     last_ping_request_ = std::chrono::steady_clock::now();
     ^~~~~~~~~~~~~~~~~~
src/esphome/components/mitsubishi_heatpump/espmhp.cpp: In member function 'void MitsubishiHeatPump::set_remote_operating_timeout_minutes(int)':
src/esphome/components/mitsubishi_heatpump/espmhp.cpp:641:5: error: 'remote_operating_timeout_' was not declared in this scope
     remote_operating_timeout_ = std::chrono::minutes(minutes);
     ^~~~~~~~~~~~~~~~~~~~~~~~~
src/esphome/components/mitsubishi_heatpump/espmhp.cpp:641:5: note: suggested alternative: 'set_remote_operating_timeout_minutes'
[... cut ...]
             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
src/esphome/components/mitsubishi_heatpump/espmhp.cpp:671:13: note: suggested alternative: 'enforce_remote_temperature_sensor_timeout'
             last_remote_temperature_sensor_update_.has_value()) {
             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
             enforce_remote_temperature_sensor_timeout
*** [.pioenvs/esp32_acmatilde/src/esphome/components/mitsubishi_heatpump/espmhp.cpp.o] Error 1
========================= [FAILED] Took 32.40 seconds =========================`

monkeypr00f, 2024-08-28:
Solved using [these](https://github.com/geoffdavis/esphome-mitsubishiheatpump/issues/157#issuecomment-2242053481) build options:

redstorm1, 2024-08-28:
> Solved using [these](https://github.com/geoffdavis/esphome-mitsubishiheatpump/issues/157#issuecomment-2242053481) build options

Confirm these build options solved the compile issue.
[image]

closed 2025-03-31 by geoffdavis