geoffdavis/esphome-mitsubishiheatpump #169: Building on Pi Pico W

sfozz, 2024-09-09:
Couple of build issues when building for this target.

I've added the following to allow the builds to get started
```diff
diff --git a/components/mitsubishi_heatpump/climate.py b/components/mitsubishi_heatpump/climate.py
index da16834..b9c4d74 100644
--- a/components/mitsubishi_heatpump/climate.py
+++ b/components/mitsubishi_heatpump/climate.py
@@ -53,6 +53,8 @@ def valid_uart(uart):
         uarts = ["UART0"]  # UART1 is tx-only
     elif CORE.is_esp32:
         uarts = ["UART0", "UART1", "UART2"]
+    elif CORE.is_rp2040:
+        uarts = ["UART0", "UART1"]
     else:
         raise NotImplementedError
```

Following this I had to add the build options from [here ](https://github.com/geoffdavis/esphome-mitsubishiheatpump/issues/157#issuecomment-2242053481) but the builds were still failing... so I added the following as suggested by `g++`
```diff
diff --git a/components/mitsubishi_heatpump/espmhp.h b/components/mitsubishi_heatpump/espmhp.h
index 110d6be..3f6ff7a 100644
--- a/components/mitsubishi_heatpump/espmhp.h
+++ b/components/mitsubishi_heatpump/espmhp.h
@@ -23,6 +23,7 @@
 #include <chrono>

 #include "HeatPump.h"
+#include <optional>

 #ifndef ESPMHP_H
 #define ESPMHP_H
```

I'm now getting the following errors, which are well outside of my c++ chops
```
src/esphome/components/mitsubishi_heatpump/espmhp.cpp: In member function 'virtual void MitsubishiHeatPump::setup()':
src/esphome/components/mitsubishi_heatpump/espmhp.cpp:703:13: error: cannot convert 'MitsubishiHeatPump::setup()::<lambda()>' to 'void (*)()'
  703 |             [this]() {
      |             ^~~~~~~~~~
      |             |
      |             MitsubishiHeatPump::setup()::<lambda()>
  704 |                 this->hpSettingsChanged();
      |                 ~~~~~~~~~~~~~~~~~~~~~~~~~~
  705 |             }
      |             ~
In file included from src/esphome/components/mitsubishi_heatpump/espmhp.h:25,
                 from src/esphome.h:23,
                 from src/esphome/components/mitsubishi_heatpump/espmhp.h:20,
                 from src/esphome/components/mitsubishi_heatpump/espmhp.cpp:21:
.piolibdeps/heatpump/HeatPump/src/HeatPump.h:40:52: note:   initializing argument 1 of 'void HeatPump::setSettingsChangedCallback(void (*)())'
   40 | #define SETTINGS_CHANGED_CALLBACK_SIGNATURE void (*settingsChangedCallback)()
      |                                             ~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~
.piolibdeps/heatpump/HeatPump/src/HeatPump.h:291:37: note: in expansion of macro 'SETTINGS_CHANGED_CALLBACK_SIGNATURE'
  291 |     void setSettingsChangedCallback(SETTINGS_CHANGED_CALLBACK_SIGNATURE);
      |                                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
src/esphome/components/mitsubishi_heatpump/espmhp.cpp:709:13: error: cannot convert 'MitsubishiHeatPump::setup()::<lambda(heatpumpStatus)>' to 'void (*)(heatpumpStatus)'
  709 |             [this](heatpumpStatus currentStatus) {
      |             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |             |
      |             MitsubishiHeatPump::setup()::<lambda(heatpumpStatus)>
  710 |                 this->hpStatusChanged(currentStatus);
      |                 ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  711 |             }
      |             ~
.piolibdeps/heatpump/HeatPump/src/HeatPump.h:41:50: note:   initializing argument 1 of 'void HeatPump::setStatusChangedCallback(void (*)(heatpumpStatus))'
   41 | #define STATUS_CHANGED_CALLBACK_SIGNATURE void (*statusChangedCallback)(heatpumpStatus newStatus)
      |                                           ~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
.piolibdeps/heatpump/HeatPump/src/HeatPump.h:292:35: note: in expansion of macro 'STATUS_CHANGED_CALLBACK_SIGNATURE'
  292 |     void setStatusChangedCallback(STATUS_CHANGED_CALLBACK_SIGNATURE);
      |                                   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```

sfozz, 2024-09-09:
```yaml
esphome:
  platformio_options:
    build_unflags:
      - "-std=gnu++11"
    build_flags:
      - "-std=gnu++17"
  name: heatpump
  friendly_name: HeatPump

rp2040:
  board: rpipicow
  framework:
    platform_version: https://github.com/maxgerhardt/platform-raspberrypi.git

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "***"

ota:
  - platform: esphome
    password: "***"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot in case wifi connection fails
  ap:
    ssid: "Heatpump Fallback Hotspot"
    password: "***"

external_components:
  - source: github://sfozz/esphome-mitsubishiheatpump

climate:
  - platform: mitsubishi_heatpump
    name: "${friendly_name}"

    horizontal_vane_select:
      name: Horizontal Vane

    hardware_uart: UART1
    baud_rate: 4800
    
```

ngist, 2025-01-04:
Man I wish i found this a few hours ago as I just went through this myself and repeated your work. Although rather than adding the #include <optional> i changed the references to std::optional in espmhp.h to esphome::optional

I made it one step further though. In the SwiCago/HeatPump library it defines `STATUS_CHANGED_CALLBACK_SIGNATURE`  based on what chip is being used. You'll want to change `#if defined(ESP8266) || defined(ESP32)` to `#if defined(ESP8266) || defined(ESP32) || defined(USE_RP2040)` [link](https://github.com/SwiCago/HeatPump/blob/dc898d5ca9579ee50e74a9e334d1a6202aeb0fe8/src/HeatPump.h#L31)

I've got a fork with the change [here](https://github.com/ngist/HeatPump), you'll need to edit https://github.com/geoffdavis/esphome-mitsubishiheatpump/blob/1feeaeb2574807a8bf5e38df6c7e9983adde0f96/components/mitsubishi_heatpump/climate.py#L155

All i can say is with that addtional change I was able to compile it and load it on the PicoW, I still need to physically connect it to a heatpump to see if it actually works.

ngist, 2025-01-16:
The above changes were enough to compile but not enough to get it to work apparently.

One more change i've identified:
`serial = HARDWARE_UART_TO_SERIAL[PLATFORM_ESP8266][config[CONF_HARDWARE_UART]]`

Needs to properly select when PLATFORM is RP2040 otherwise the wrong serial will be selected.

Here's my attempt though it seems to have stopped me from doing wireless updates after making this change.
```    
if CORE.is_rp2040:
    serial = HARDWARE_UART_TO_SERIAL[PLATFORM_RP2040][config[CONF_HARDWARE_UART]]
else:
    serial = HARDWARE_UART_TO_SERIAL[PLATFORM_ESP8266][config[CONF_HARDWARE_UART]]
```

See here:
https://github.com/geoffdavis/esphome-mitsubishiheatpump/blob/1feeaeb2574807a8bf5e38df6c7e9983adde0f96/components/mitsubishi_heatpump/climate.py#L101

https://github.com/esphome/esphome/blob/b4a804cc779bcb8a3b1480a44741bf9f29caa33c/esphome/components/logger/__init__.py#L111

sfozz, 2025-01-16:
@ngist thanks for looking into this... I'll have a look this weekend and update here!

ngist, 2025-01-16:
No problem, one other thing worth mentioning before you hook this up to your unit. Do you have an interface circuit for the PicoW? As I understand it the CN105 uses 5v for the serial lines for the PicoW-TX->CN105-RX connection you should be able to wire it directly no need to step up the voltage as the output level is above the standard Vih 2v threshold, same with the 5V power and ground. **For the CN105-TX->PicoW-RX, you'll want something to step down the voltage from 5V to 3V** a resistor divider, diode solution, level shifter or whatever.

sfozz, 2025-02-06:
Thanks for the heads up... I'd already added one 😇 

[image]

sfozz, 2025-02-11:
@ngist which version of esphome are you using?

sfozz, 2025-02-11:
getting closer... this is connected to the heatpump but isn't communicating correctly with it

```
INFO ESPHome 2024.12.4
INFO Reading configuration /config/heatpump.yaml...
INFO Starting log output from 192.168.100.10 using esphome API
INFO Successfully connected to heatpump @ 192.168.100.10 in 0.004s
INFO Successful handshake with heatpump @ 192.168.100.10 in 0.225s
[22:26:19][I][app:100]: ESPHome version 2024.12.4 compiled on Feb 11 2025, 21:55:46
[22:26:19][C][wifi:600]: WiFi:
[22:26:19][C][wifi:428]:   Local MAC: 28:CD:C1:11:2C:9B
[22:26:19][C][wifi:433]:   SSID: [redacted]
[22:26:19][C][wifi:436]:   IP Address: 192.168.100.10
[22:26:19][C][wifi:439]:   BSSID: [redacted]
[22:26:19][C][wifi:441]:   Hostname: 'heatpump'
[22:26:19][C][wifi:443]:   Signal strength: -63 dB ▂▄▆█
[22:26:19][C][wifi:447]:   Channel: 6
[22:26:19][C][wifi:448]:   Subnet: 255.255.255.0
[22:26:19][C][wifi:449]:   Gateway: 192.168.100.1
[22:26:19][C][wifi:450]:   DNS1: 192.168.100.1
[22:26:19][C][wifi:451]:   DNS2: 0.0.0.0
[22:26:19][C][logger:185]: Logger:
[22:26:19][C][logger:186]:   Level: DEBUG
[22:26:19][C][logger:188]:   Log Baud Rate: 115200
[22:26:19][C][logger:189]:   Hardware UART: USB_CDC
[22:26:19][I][MitsubishiHeatPump:074]: ESPHome MitsubishiHeatPump version 2.5.0
[22:26:19][I][MitsubishiHeatPump:771]:   Supports HEAT: YES
[22:26:19][I][MitsubishiHeatPump:772]:   Supports COOL: YES
[22:26:19][I][MitsubishiHeatPump:773]:   Supports AWAY mode: NO
[22:26:19][I][MitsubishiHeatPump:774]:   Saved heat: -1.0
[22:26:19][I][MitsubishiHeatPump:775]:   Saved cool: -1.0
[22:26:19][I][MitsubishiHeatPump:776]:   Saved auto: -1.0
[22:26:19][E][component:082]:   Component esphome.coroutine is marked FAILED
[22:26:19][C][mdns:116]: mDNS:
[22:26:19][C][mdns:117]:   Hostname: heatpump
[22:26:19][C][esphome.ota:073]: Over-The-Air updates:
[22:26:19][C][esphome.ota:074]:   Address: heatpump.local:2040
[22:26:19][C][esphome.ota:075]:   Version: 2
[22:26:19][C][esphome.ota:078]:   Password configured
[22:26:19][C][safe_mode:018]: Safe Mode:
[22:26:19][C][safe_mode:019]:   Boot considered successful after 60 seconds
[22:26:19][C][safe_mode:021]:   Invoke after 10 boot attempts
[22:26:19][C][safe_mode:022]:   Remain in safe mode for 300 seconds
[22:26:19][C][api:140]: API Server:
[22:26:19][C][api:141]:   Address: heatpump.local:6053
[22:26:19][C][api:143]:   Using noise encryption: YES
[22:26:42][I][safe_mode:041]: Boot seems successful; resetting boot loop counter
[22:26:42][D][rp2040.preferences:117]: Saving preferences to flash...
```

sfozz, 2025-02-14:
Had to break out the logic analyzer, but even if I set the tx and rx pins to 6 and 7 the UART1 signals are sent over 11 and 12.

any suggestions before I re-solder and try again?

sfozz, 2025-02-15:
So I tried a number of different ways to use the alternative pins for UART1... no dice!

So I have re-soldered to use the default pins... Winner!

ngist, 2025-03-13:
Do you mind sharing your final config? It sounds like you didn't need any other code changes?

I'm just starting back on trying to get mine to work.

ngist, 2025-03-13:
I took a look through your fork, I see you changed the ESP_LOGCONFIG() calls to EXP_LOGD() which I was about to try, because i wasn't seeing debug messages from setup() so instead I've just set my config to use your fork. 

All i get is an output very similar to the one you posted above with the core result of:
```
[21:35:02][I][MitsubishiHeatPump:074]: ESPHome MitsubishiHeatPump version 2.5.0
[21:35:02][I][MitsubishiHeatPump:794]:   Supports HEAT: YES
[21:35:02][I][MitsubishiHeatPump:795]:   Supports COOL: YES
[21:35:02][I][MitsubishiHeatPump:796]:   Supports AWAY mode: NO
[21:35:02][I][MitsubishiHeatPump:797]:   Saved heat: -1.0
[21:35:02][I][MitsubishiHeatPump:798]:   Saved cool: -1.0
[21:35:02][I][MitsubishiHeatPump:799]:   Saved auto: -1.0
[21:35:02][E][component:082]:   Component esphome.coroutine is marked FAILED]
```

but there are no other debug messages related to the setup function that I would expect to see which makes me think it's crashing on construction before setup() is called or something is blocking it from being published after the setup fails.

ngist, 2025-03-13:
To answer your questions regarding non-default pins, the SwiCago/Heatpump library ignores the rx/tx pin settings unless ESP32 is defined. I just made a commit to my fork to make the behavior the same for RP2024 and ESP32 to attempt to allow custom pins, whether it works or not I do not know as I can't get mine to work at all yet.

I'm currently trying to use UART0 with pins 1/2 so maybe I'll resolder tomorrow and try UART1 with pins 6/7.

sfozz, 2025-03-13:
This is the esphome config

```
esphome:
  platformio_options:
    build_unflags:
      - "-std=gnu++11"
    build_flags:
      - "-std=gnu++17"
  name: heatpump

rp2040:
  board: rpipicow

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "**"

ota:
  - platform: esphome
    password: "**"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot in case wifi connection fails
  ap:
    ssid: "Heatpump Fallback Hotspot"
    password: "**"

external_components:
  - source:
      type: git
      url: https://github.com/sfozz/esphome-mitsubishiheatpump
      ref: build_9
    refresh: 10s

climate:
  - platform: mitsubishi_heatpump
    name: HVAC
    hardware_uart: UART1
    baud_rate: 2400

    visual:
      min_temperature: 16
      max_temperature: 31
      temperature_step: 1.0
# Example configuration entry
button:
  - platform: restart
    name: "Restart"
```

ngist, 2025-03-30:
Thanks @sfozz it turns out my issue was baudrate being too high. FYI this project seems mostly dead, however https://github.com/echavet/MitsubishiCN105ESPHome seems to be very active. I got a PR landed recently that allows it to work on the Pico W so it should be plug and play.

closed 2025-03-31 by geoffdavis