geoffdavis/esphome-mitsubishiheatpump #171: ESP32 Unable to build file

nchlstoo, 2024-10-04:
Hi

I have an ESP32 Dev Kit V1 and tried to compile the build and it gave me these errors. 

[logs_test_run.txt](https://github.com/user-attachments/files/17261868/logs_test_run.txt)

This is my build:

`esphome:
  name: test
  friendly_name: test

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable Home Assistant API
api:
  encryption:
    key: "YupWfhldRcwCT6Af6NdlqWww62j8fettiH2RB0Ff/1M="

ota:
  - platform: esphome
    password: "b5e44f86f4bab4632c522642441ae1eb"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Test Fallback Hotspot"
    password: "4mdv0cIT7nc1"

# Enable Web server.
web_server:
  port: 80

captive_portal:

# Sync time with Home Assistant.
time:
  - platform: homeassistant
    id: homeassistant_time

sensor:
  - platform: wifi_signal
    name: "$devicename WiFi Signal"
    update_interval: 60s
  - platform: uptime
    name: "$devicename Uptime"

external_components:
   - source: github://geoffdavis/esphome-mitsubishiheatpump

climate:
  - platform: mitsubishi_heatpump
    name: "${devicename}"
    update_interval: 500ms

    # ESP32 only - change UART0 to UART1 or UART2 and remove the
    # logging:baud_rate above to allow the built-in UART0 to function for
    # logging.
    hardware_uart: UART0
    supports:
      mode: ['HEAT_COOL', 'COOL', 'DRY', 'FAN_ONLY']
      fan_mode: ['AUTO', 'LOW', 'MEDIUM', 'MIDDLE', 'HIGH']
      swing_mode: ['OFF', 'VERTICAL']
    visual:
      min_temperature: 16
      max_temperature: 31
      temperature_step: 1.0

text_sensor:
  - platform: version
    name: "$devicename ESPHome Version"
  - platform: wifi_info
    ip_address:
      name: ${devicename} IP
    ssid:
      name: ${devicename} SSID
    bssid:
      name: ${devicename} BSSID`

unregist, 2024-10-17:
Duplicate of #117 

Add this to your config file to fix:
```
esphome:
  platformio_options:
    build_unflags:
      - "-std=gnu++11"
    build_flags:
      - "-std=gnu++17"
```

ptbpa, 2024-12-29:
I had the same issue and this solved it! thank you very much!

> Duplicate of #117
> 
> Add this to your config file to fix:
> 
> ```
> esphome:
>   platformio_options:
>     build_unflags:
>       - "-std=gnu++11"
>     build_flags:
>       - "-std=gnu++17"
> ```

closed 2025-03-31 by geoffdavis