geoffdavis/esphome-mitsubishiheatpump #174: Missing vane info and control

haraldhh, 2025-01-17:
I have a Mitsubishi RW-25 connected to a D1 mini and esphome-mitsubishiheatpump. 

I am missing vane control and info. Is this possible with esphome-mitsubishiheatpump or have I misconfigured it somehow?

DamijanD, 2025-01-17:
vane is only on internal AC unit, right? This is not supported by this component it is for heat pumps only (air 2 water)

haraldhh, 2025-01-17:
The documentation has this section in examples;

    horizontal_vane_select:
      name: Horizontal Vane
    vertical_vane_select:
      name: Vertical Vane

I do think this is no only for air2water but also air to air. The list of supported units is mostly air to air pumps. 

I tried to add that previously mentioned section to my config, but it didn't expose any vane_* in the entity.

DamijanD, 2025-01-17:
Uf I am so so sorry - I though that I am on the HP repository (https://github.com/gekkekoe/esphome-ecodan-hp)

domerich, 2025-01-25:
I have the same questions, with other integrations I can set up down 1,2,3,4,5 but here not. 

this is an annoying work around:

horizontal_vane_select:
  name: Horizontal Vane
vertical_vane_select:
  name: Vertical Vane

how can it be integrated in the climate control panel of HomeAssistant?

Injineers, 2025-02-18:
like this?

[image]

I also want to implement this ...that screen shot is from the mitsubishi2mqtt i am running on 1 of my units, but am rolling over to this one instead (esphome ver)

but this is something I know my wife is going to want before I do this. Might have to pick through the other library & figure how they do it

domerich, 2025-02-19:
Yes like this, ideally both for horizontal and vertical swing