geoffdavis/esphome-mitsubishiheatpump #177: Cannot update

Aresitoo, 2025-02-19:
Hi I am getting this error since last update of ESPHOME and cant figure out what to replace in the first place of the code 

`INFO ESPHome 2025.2.0
INFO Reading configuration /config/esphome/esphome-web-fd4540.yaml...
Failed config

esphome: [source /config/esphome/esphome-web-fd4540.yaml:2]
  name: esphome-web-fd4540
  
  Please remove the `platform` key from the [esphome] block and use the correct platform component. This style of configuration has now been removed.
  platform: ESP8266
  board: esp01_1m`

This is my conig file that was working fine : 

`esphome:
  name: esphome-web-fd4540
  platform: ESP8266
  board: esp01_1m

external_components:
  - source: github://geoffdavis/esphome-mitsubishiheatpump

substitutions:
  devicename: mitsubishi-aa-atico #EJEMPLO
  upper_devicename: Aire atico #EJEMPLO

logger:
  baud_rate: 0 #DESACTIVADO PARA NO CAUSAR CONFLITOS

# Enable Home Assistant API.
api:
  encryption:
    key: "5K4si2P8fdiyPEA3Jo+8vtKq3b/Xea0R9PheYuvqOf8=" #OPCIONAL

# Enable over-the-air updates.
ota:
    platform: esphome
    password: !secret ota_password

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "removed for pasting here"
    password: "removed to paste it here"

climate:
  #temp: 17-28; no diffuses, no vane
  - platform: mitsubishi_heatpump
    name: $upper_devicename
    hardware_uart: UART0
    baud_rate: 9600
    supports:
      mode: [DRY, COOL, HEAT, FAN_ONLY]
      fan_mode: [LOW, HIGH]
      swing_mode: []
    visual:
      min_temperature: 17
      max_temperature: 28
      temperature_step: 1.0`

Aresitoo, 2025-02-19:
Sorry for the nuisance, I asked Gemeni to correct my code and it works now, solution was : 

esphome:
  name: esphome-web-fd4540
  # platform: ESP8266  <--- REMOVE THIS LINE
  # board: esp01_1m    <--- REMOVE THIS LINE

esp8266:           # <--- ADD THIS PLATFORM COMPONENT BLOCK
  board: esp01_1m    # <--- MOVE BOARD HERE

closed 2025-02-19 by Aresitoo

xbmcnut, 2025-02-20:
Corrected code now looks like this to build properly under 2025.2.0
```
esphome:
  name: ${name}
  friendly_name: Spare Room Heat Pump  
  comment: Custom D1 mini inside aircon in guest room
  # Boards tested: ESP-01S (ESP8266), Wemos D1 Mini (ESP8266); ESP32 Wifi-DevKit2
  platformio_options:
    build_flags: 
      - -std=gnu++17
    build_unflags:
      - -std=gnu++11

esp8266:
  board: d1_mini
```